nailbuster/esp8266FTPServer #2: Folder emulation

holgerlembke, 2016-02-09:
Hi David,

I just stumbled upon this nice ftp server, works great out of the box.

Is any work done on emulating folders?

nailbuster, 2016-02-09:
good to hear its working....

nope, no emulating folders yet....if/when the spiffs supports directories 'fully', then it's something to look at....

pgScorpio, 2016-06-20:
Would it be that hard to simulate folders from the filenames ??

Since the filenames contain slashes you know which part of the filename is a path and which part is the filename....

Now, , if you have "folders", because the filenames contain slaches you can't up- or download them via ftp.

holgerlembke, 2016-06-20:
It could be done but it is a lot of detail work. Let me pick three:

- for "create folder" you have somehow manage empty folders. Could be a dummy file ".". Easy.

- for all folder related functions (remove, rename, make) you always have to iterate over all (flat) files to collect/check/work your folder tree.

- mixing folder walk (find first/next) and delete/insert files and folders the same time (like creating or deleting files in a find-first-next-loop over a folder) must somehow be a defined and be consistent.

Does not sound terrible difficult. But as said before, it is a job for the file system, not the ftp server.

pgScorpio, 2016-06-20:
I agree it should be done on the filesystem !
I really don't understand why they (still) did not implement it......

If I can find the time ill look into it myself..