nailbuster/esp8266FTPServer #3: ESP8266FtpServer not working!

Ariel-International, 2016-03-08:
I successfully used this library in a project, but suddenly nothing seems to work anymore!
If I set the ESP core to 1.6.5 there is no SPIFFS.
If I set either 2.0 or 2.1 I get an error in buf format for ReadByte function in the library.

As it looks, the standard is now more strict in the compiler, so it won't accept conversion from char to unsigned char.

I reinstalled the Arduino IDE (1.6.7 I admit) to no avail.
Now I removed all references to FtpServer from my projects, but as soon as I include the library the sketch compiles but hangs on startup.
Is it possible to give a fast check to this issue?
Shouldn't be a big problem for somebody who knows where to look.

nailbuster, 2016-03-08:
I just checked with 1.6.5 and v2.1 esp Library and everything works fine....you should stick with 1.6.5.

All i can suggest,  perhaps format the spiffs?  make sure you don't have multiple versions of the esp library around the drive (sometimes compiler will use wrong one)...  

try just the ftpserversample from my library and see if that works....

let me know.

tzapu, 2016-03-11:
i am using it with arduino 1.6.7, latest github of the esp8266 core for arduino and it all works, provided the client is ok (filezilla seems fine, cyberduck no)

Ariel-International, 2016-03-11:
Hello, I checked everything and came up with the following:
- ftpserversample works with 1.6.7
- I found another major issue in my sketch related to free memory size
- I kept having random problems with SPIFFS until
- I read an article saying that an include for FS.h should be put in the first line of the code, before anything else.
Done that, I am now moving forward without problems.
I don't know if that helped or it was by chance, this was my observation.
With Filezilla!

closed 2016-03-11 by Ariel-International

tzapu, 2016-03-11:
i can confirm i had to include FS.h on the first line, otherwise unpredictable crap happened

RTrioux, 2020-01-03:
Thank you @Ariel-International  and @tzapu , FS.h fixed my issue :)