nailbuster/esp8266FTPServer #5: dataIp=softAPIP if in AP mode

digimeas, 2016-04-20:
Hi David.  Thanks for esp8266FTPServer!

My ESP is set up in AP mode so doesn't have WiFi.localIP set.  To connect from my android phone I added: 
```c++
  else if( ! strcmp( command, "PASV" ))
  {
    if (data.connected()) data.stop();
    dataIp = WiFi.localIP();
    if(0 == dataIp[0])            ///  <<<  These two lines (There's surely a better test !?)
      dataIp = WiFi.softAPIP();   ///  <<<
```
Don't know if it's the right way but it works in this instance. :-)

Thanks again
Doug

nailbuster, 2016-04-21:
thanks,  your fix/solution seems simple enough...I will implement soon.

pgScorpio, 2016-06-20:
I Used:

```
boolean FtpServer::processCommand()
{
 .........
 .........

 else if( ! strcmp( command, "PASV" ))
  {
    if (data.connected()) data.stop();
    dataIp = client.localIP();              /// <<< use the clients local IP() 

```
Even simpler, and more correct.

closed 2017-10-21 by nailbuster