nailbuster/esp8266FTPServer #7: Insufficient info to get this to compile

boylesg, 2017-01-01:
I have unzipped your library along with the one at https://github.com/esp8266/Arduino.
It still won't't compile.
Arduino IDE  complains about not being able to find queue.h - I found this in tools/sdk/.... and copied it to C:\Program Files (x86)\Arduino\libraries\esp8266FTPServer and I got past this compile error.
Now it is complaining about not finding <functional> and I can't find this file in your github repository or the one above.
So again, clearly your library has other dependencies that you are forgetting about and that I don't have in my arduino installation.

nailbuster, 2017-01-01:
you need to take a step back and get your IDE/esp8266 core working first....

Can you compile the esp8266 sample "Hello Server" and upload it to your esp?

Once you have that working then getting this library to work is just as simple....

btw:  it sounds like you haven't set the proper board in your IDE before compile...

boylesg, 2017-01-02:
Well that is exactly the sort of thing you need to detail David – it is not remotely clear from the little doco you have provided. 

I am trying to compile and upload to a mega with an ESP8266 shield, not to a standalone ESP8266 board.

 

You also need to provide a link to that other repository you pointed out to me and make it clear to people that they need to install that as well.

Otherwise it is like trying to find a needle in a haystack of needles – there are dozens of different ESP8266 repositories. 

Which one have you used – who the hell can possibly know?

 

And also a link to how you get the ESP8266 sample compiling and working – you must have used someone else’s instructions when you were setting this up yourself the first time, so provide the link that you used.

 

Neither of us are chinese who can’t speak English and can’t write a decent manual for shit. And manuals don’t need to be enormous documents – just a series of short and clear step by step instructions in a numbered list. It would take half an hour tops in your case – tiny a fraction of the amount of time you have spent coding your library. I always create my manuals with screen shots to make it even clearer to newbies.

 

Gregary Boyles

 

Greg's Indigenous Plants & Landscapes

 

ABN: 65 285 170 251

 

Phone: 94013696     Mobile: 0414318470

 

Email: greg@gregsindigenouslandscapes.com.au

 

Mail Order Nursery: http://www.gregsindigenouslandscapes.com.au/OnlineNursery.php

 

Website: http://www.gregsindigenouslandscapes.com.au

 

"The exotic vegetation that replaces indigenous plant communities in urbanising regions, disassociates us from the rhythms and diversity of the native landscape and a sense of the place; and we are the poorer because of it." 

 

Professor of Landscape Architecture, York University, Canada