nailbuster/esp8266FTPServer #8: Does not work with ESPAsyncWebServer

kdobrev, 2017-04-07:
Hi, I would like to use the ftp server, but I am using ESPAsyncWebServer. The ftp tries to connect but then stops and does no do anything or issues a random error (username, password or something else).

The ESPAsyncWebServer does not use server.handleClient(); in the loop(). Do have any idea how can be done so that both work together?

nailbuster, 2017-04-07:
I doubt it would work with ASyncWebServer.  Don't know if you can mix both types in the same project.

Are you sure you put ftpServer.handleFTP();  in your loop?

if you want checkout my AsyncWebServer project that has integrated FileBrowser(upload/delete to spiffs)... doesn't use ftp though.  https://github.com/nailbuster/myWebServerAsync

kdobrev, 2017-04-07:
Yes, I have the ftpSrv.handleFTP(); My goal is to get data out SPIFFS.

nailbuster, 2017-04-07:
Try out my WebServerProject....  Once started you just go to http://192.168.1.xx/browse    that will bring up an html SPIFFs browser/upload/download/delete

pdfelen, 2017-08-04:
For me it works, but on heavy traffic there will be multiple retries. The same with normal WiFi client.
Rewriting WiFi client to Async client should not be such a big job....
Let's see....

pdfelen, 2017-09-22:
Update:
Started over with another concept. Might be interesting...  Keep you posted.