nailbuster/esp8266FTPServer #9: Faster transfers

bbx10, 2017-06-30:
I made some changes to increase transfer speeds to the range of 100 - 300 Kbytes/s. Also reduces blocking so ftp can co-exist with other services. Let me know if you want a PR.

https://github.com/bbx10/esp8266FTPServer/tree/speedup

nailbuster, 2017-06-30:
great work!  thanks.  Good timing too, as I just got a esp32 and was going to test the library with it.  I'll test out your changes and verify esp32 library as well.  (unless someone has already tried it on esp32 and can confirm it works fine...)

bbx10, 2017-06-30:
SPIFFs for arduino-esp32 is not ready but I will also try it when it is.

bbx10, 2017-06-30:
The FTP server is faster than expected. ` wget ftp://esp8266:esp8266@192.168.xxx.yyy/512kb.dat` shows 877 KB/s for a 512 KB file.

```
$ wget ftp://esp8266:esp8266@192.168.1.129/512kb.dat
--2017-06-30 11:40:08--  ftp://esp8266:*password*@192.168.1.129/512kb.dat
           => ‘512kb.dat.6’
Connecting to 192.168.1.129:21... connected.
Logging in as esp8266 ... Logged in!
==> SYST ... 
Server error, can't determine system type.
==> PWD ... done.
==> TYPE I ... done.  ==> CWD not needed.
==> SIZE 512kb.dat ... 526294
==> PASV ... done.    ==> RETR 512kb.dat ... done.
Length: 526294 (514K) (unauthoritative)

100%[======================================>] 526,294      877KB/s   in 0.6s   
```

capedra, 2017-07-29:
@bbx10 
Congratulations !!!

capedra, 2017-07-30:
Well, I'm doing some going to test on 3 MB of SPIFFS now, because with 14900866 total bytes of flash, every now and then my files get corrupted after transferring them to SPIFFS. I wish to have some luck this time. You can get some updates on here:

https://github.com/esp8266/Arduino/issues/2581

beegee-tokyo, 2017-10-16:
@nailbuster & @bbx10 
As SPIFFS on ESP32 is working now I added support for the ESP32 to your FTP server.
I merged as well bbx10's changes for speed improvement and reduction for blocking.
Tested on my ESP32 dev module with FileZilla.

I added as well a library.json (needs to be adapted though) so that the library can be used with PlatformIO.

See pull request [Added ESP32 support, merged bbx10 speed improvement](https://github.com/nailbuster/esp8266FTPServer/pull/11)