nailbuster/esp8266FTPServer #12: Large File Upload

nailbuster, 2017-10-18:
I've been testing the changes for faster upload/download and esp32 support.

I've found almost everything works well with both esp8266 and esp32....

Found one issue that don't know if others can replicate.  Upload of files larger than 512K (524,288) bytes will 'stall'.  On the esp8266 the file will eventually upload fine,  on the esp32 sometimes the file will fail upload but trying again usually gets the file to the esp32.

downloading files >512K works fine and fast.

all testing was done using FileZilla.

esp32 itself isn't 'perfect' yet...  I was testing it hard and got to the point where I had to reformat the spiffs to get it working but couldn't replicate it.  

thanks to the all those who have contributed/helped.

beegee-tokyo, 2017-10-20:
I can confirm the problems with uploading files larger than 512Kb on ESP32. I guess the SPIFFS library still has bugs.

clydde, 2019-05-09:
Hello, I am looking to upload a 180kb text file and I do not get it, the WinSCP is uploaded when uploading 142kb.
Any advice?

clydde, 2019-05-09:
Auto Answer:
For esp8266 better use WinSCP
For ESP32 Filezilla it's ok!