nailbuster/esp8266FTPServer #14: Failed to connect with Filezilla

Ahmedkdk, 2017-11-28:
Hello nailbuster,

Firstly, thank you for the library. 
I tried your example, I editted in my ssid and password and keyed in the user name and password from my Filezilla site in the main function of the example. and I couldnt find a place to key in the Host IP address. I ran the code on my ESP8266 Huzzah Feather, and it only shows this:

Connected to MySSID
IP address: 192.168.179.8
SPIFFS opened!

then it just gets stuck there, I am sure its a beginner's inquiry since I am not familiar with FTP servers and all, seeking your assistance

Regards

tobozo, 2017-11-28:
From the [ReadMe](https://github.com/nailbuster/esp8266FTPServer/blob/master/README.md): 

```
You need to setup Filezilla(or other client) to only allow 1 connection.. To force FileZilla to use the primary connection for data transfers: Go to File/Site Manager then select your site. In Transfer Settings, check "Limit number of simultaneous connections" and set the maximum to 1

only supports Passive ftp mode....
```

Ahmedkdk, 2017-12-06:
It worked! Thank you very much sir!

closed 2017-12-12 by Ahmedkdk