nailbuster/esp8266FTPServer #23: rewrite to use async TCP-Functions

Monarch73, 2018-11-21:
Hi.

I had troubles with your FTP-Server uploading small files to SPIFFS (ie. 300byte files).
I noticed the troubles where coming from the blocking TCP-Functions that you use. It makes it behave very unstable.

I have rewritten the entire thing and made use Async TCP-Library from me-no-dev. Its way more responsive and works very nicely. I wanted to share the code with you.

https://github.com/Monarch73/MicroHomebridgeAlexaEsp8266/blob/master/MicroHomebridgeAlexaEsp8266/FtpServ.h

Its just this header-file that also includes the implementation. For me it's part of a different project. Maybe you want to convert it to a stand-alone project, if you like.

ersivv, 2019-03-03:
@Monarch73 not work write file to SPIFFS

Monarch73, 2019-03-03:
@ersivv no, **work** write file to SPIFFS

ersivv, 2019-03-04:
indeed, the Winscp client has written a file