nailbuster/esp8266FTPServer #24: esp8622 crash..

wchpikus, 2018-11-21:
esp with working wifi, www server, spiffs, socket, oled via spi.
added ftp, and when transfering file esp crash.
using filezilla, disconnect client when choiced transfer file (i have to connect again, wait for connection).

Monarch73, 2018-11-22:
@wchpikus 
do you mind trying out my ESP-FTP-Server? Its a single .h-File that you can copy and paste into your project. It just depends on the AsyncTCP-Library from @me-no-dev

Please not that it is required that you limit the number of concurrent connections to 1 in filezilla. Filetransfers are done via PORT-Command, so it supports non-passive mode only.

https://github.com/Monarch73/MicroHomebridgeAlexaEsp8266/blob/master/MicroHomebridgeAlexaEsp8266/FtpServ.h

wchpikus, 2018-11-22:
hi
yes, i import your library to my project.
ESP8266FtpServer.h and ESP8266FtpServer.cpp files, and setup in main and setup in ino.
connections is only1 in filezilla and non-passive mode.

I can remove, see dir and rename, but upload made crash:(
you write about this file FtpServ.h?
I have to add this file also?

wchpikus, 2018-11-22:
yes, i corrected post:)

now i will thy our ftp.

Could you send me some more info about that?

e.g how to use?

FtpServ(char *username, char *password) = FtpServ(test,test); i see, but do you need inicialize or handle something?

Monarch73, 2018-11-22:
```
FtpServ *srv = new FtpServ("username","password");
srv->begin();
```
Put that code somewhere in the setup()
Note that for now username and password is ignorened when you login. Security was not my concern, when I created this...it'll simply accept any user/pass you give it.

wchpikus, 2018-11-22:
https://github.com/me-no-dev/ESPAsyncTCP/tree/master/src
can i get it ESPAsyncTCP.h and ESPAsyncTCP.cpp from this link?

Monarch73, 2018-11-22:
https://github.com/me-no-dev/ESPAsyncTCP/archive/master.zip

download this zip and in arduino studio you click "sketch" -> "library" -> add from zzip

wchpikus, 2018-11-22:
ok i now test it..
is it works with spiffs?

wchpikus, 2018-11-22:
Spiifs ready, 4M ( 3M spiffs for data).

Monarch73, 2018-11-22:
Hey, at least you see no crashes :-). But I just realized forgot to meantion a few more things:

- PASV is not supported. Use PORT instead (non-passive-mode only). Use the connection manager in Filezilla to configure this. There you also limit the number of concurrent connections to 1
- Subdirectories in spiffs are not supported.. You can only up/download to main directory on ESP
- Length of filenames is limited to 32 characters in spiffs...

Monarch73, 2018-11-22:
also please make sure, that you don't use any connection security features..
SIZE and MDTM are not supported...(and not required)

wchpikus, 2018-11-22:
ok setuped to activ:

Status:	Connecting to 192.168.11.1:21...
Status:	Connection established, waiting for welcome message...
Status:	Server does not support non-ASCII characters.
Status:	Logged in
Status:	Retrieving directory listing...
Status:	Directory listing of "/" successful
Status:	Disconnected from server
Status:	Connecting to 192.168.11.1:21...
Status:	Connection established, waiting for welcome message...
Status:	Server does not support non-ASCII characters.
Status:	Logged in
Status:	Starting upload of G:\oled animation\OLED-esp-aux_phone_english_www_spiffs_ftp_v6\c8_d7_b0_1a_34_e4.txt
Status:	File transfer successful, transferred 17 140 bytes in 1 second
Status:	Retrieving directory listing of "/"...
Status:	Directory listing of "/" successfu

ok but when i send 160kb file...crash:)

dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

 dataconnection received data from 192.168.11.2 

Exception (0):
epc1=0x4020ab3c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffff790 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffff930:  4000050c 07672d1f 401021f2 00000100  
3ffff940:  400043df 00000030 3fffc248 4000050c  
3ffff950:  40004225 00000030 00000014 ffffffff  
[... 77 lines cut ...]
3ffffe30:  3fff40b4 00000218 3fff496c 0000ff00  
3ffffe40:  3fff16c0 0000038f 0000038f 00000000  
3ffffe50:  3fff5894 00000218 3fff4f2c 402194c5  
3ffffe60:  2e323931 2e383631 3fff5b8a 4021a4d1  
3ffffe70:  3fff5894 00000218 3fff5b8a 4020ab14  
3ffffe80:  00000000 00000000 00000000 401071f0  
3ffffe90:  020ba8c0 00000008 3ffef928 00000030  
3ffffea0:  3fff5894 3fff5b44 00000000 4021a096  
3ffffeb0:  3fff0000 00000000 3fff2a4c 40215e50  
3ffffec0:  3fff5974 00000001 3ffeab24 00000064  
3ffffed0:  00000860 3ffea988 3ffea988 3fff2a6c  
3ffffee0:  3fff5974 3fff2a50 3fff2a4c 40215e94  
3ffffef0:  3fff5974 3fff2a50 3fff2a51 40223cd4  
3fffff00:  010ba8c0 0000000a 3ffea988 00000000  
3fffff10:  00000000 0000024e 3fffff80 3fff5b62  
3fffff20:  3fff2bf0 3fff5b44 3fff2cd8 40227c01  
3fffff30:  00000014 0000038f 0000038f 3fff2bf0  
3fffff40:  3fffdc80 3fff3194 3fff496c 3fff3214  
3fffff50:  00000008 3fff2bf0 3fff5b44 40221259  
3fffff60:  3fffdc80 3fff3194 3fff496c 4010453c  
3fffff70:  4023e7fe 3fff3194 3fff496c 4023e810  
3fffff80:  3fff5b54 3fff5b44 00000000 3fffdcb0  
3fffff90:  3ffeffd8 00000000 3fff496c 40240773  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

Monarch73, 2018-11-22:
I don't think that crash is related to the ftp-server code. It must be something in the sdk. Do you know what version of sdk you use?

wchpikus, 2018-11-22:
How to check it in arduino?

wchpikus, 2018-11-22:
AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1

Monarch73, 2018-11-22:
2.4.1 That should do....even thou you should be able to update to 2.4.2...but I don't think that will fix the problem. Let me check when I am back at my desk...I must admin that I never tested this with files >100k

Monarch73, 2018-11-22:
In the meantime, you can try to reformat spiffs...might be corrupted or something...

wchpikus, 2018-11-22:
I formated few times..it is no flash problem;)

Please try it and write..and what sdk have you.

Monarch73, 2018-11-22:
I always work with the latest release, so its 2.4.2 for me.... But I also work with Visual Studio Community with VMicro plugin, so I would not know how you can update your Arduino Studio....

Monarch73, 2018-11-22:
another thing you can try: Create a new project with just the ftpServer-code in it. This way you make sure that no other parts of your code are causing the crash...

wchpikus, 2018-11-22:
give me some time, i have to update and install new release:)

wchpikus, 2018-11-22:
ng core 'esp8266' from platform in folder: AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2

ok updated, i test it now...

Monarch73, 2018-11-22:
I have bad news for you. I just uploaded a 300KB File to my nodemcu-Module with no problems...

I used the following code

```
FtpServ *ftpServer;
void setup() {
	Serial.begin(115200);
	delay(20);
	SPIFFS.begin();
	// create access point
	WiFi.mode(WIFI_STA);
	WiFi.begin("xxxx", "xxxx");

	while (WiFi.status() != WL_CONNECTED) {
		delay(500);
		Serial.print(".");
	}

	Serial.println("IP address: ");
	Serial.println(WiFi.localIP());
	ftpServer = new FtpServ("1", "2");
	ftpServer->begin();
}

// the loop function runs over and over again until power down or reset
void loop() {
  
}
```

wchpikus, 2018-11-22:
now i disabling all process to eliminate problem..
please wait:)

Monarch73, 2018-11-22:
I just noticed that ftp-code does not do a SPIFFS.begin()....that needs to be done in the setup()-function as well

wchpikus, 2018-11-22:
SPIFFS.begin() I have..
This is not a problem..some of my function destruct stack..
i have find what..
now i'll test your code.

wchpikus, 2018-11-22:
hmm,  you have right...your code works well, so i have to find what do not want to work with this..
i thing this is connected with #<WiFiClient.h> and client.connect(gateway,httpPort) (i listening on 23 port from other device)..
i try to disable it when using ftp..

wchpikus, 2018-11-22:
Did you test of size of file after and before upload?
I have some strange value

Monarch73, 2018-11-22:
Nope...the code does not check filesizes...its working fine for me, too...

also, having multiple connections should generally not be an issue.  Maybe you want to try a different board?

wchpikus, 2018-11-22:
Only your code works on the same board, so this is isue of rest of my firmware..but i will check it all, disable one by one..

wchpikus, 2018-11-22:
i see, but you know, i have oled spi,telnet (two ways),and it could be no time, or interrupt some function, when receive data and got some data from other socket (telnet).
i check it, i have lot of work with them, but i see your code works..
Mayby this first also works, it must be clear, for future:)

wchpikus, 2018-11-22:
did you see difference between generic and nodemcu?

Monarch73, 2018-11-22:
what you mean "generic"? What board do you have? (amazon-link)

wchpikus, 2018-11-22:
nodemcu (aliexpress)
i write about compile to generic esp module vc nodemcu.

wchpikus, 2018-11-22:
i disable all function from loop, still crash.
could you (if you have experience, in setup i connect client (client.connect(gateway,httpPort) ) to 23 port.
and do not disconnect them..
your ftp make new connection to port 21 to client or different port?
i thing this is problem, nothing more i do not have run when client connecting to ftp server.

Monarch73, 2018-11-22:
where is "FtpServ * srv=new FtpServ(...)"-line  in your code? Inside  setup() or outside? Try to move the declaration in a global scope-right below the #include-lines.....

Monarch73, 2018-11-22:
I can't believe that the port-number could cause these kind of problems....

wchpikus, 2018-11-22:
where is "FtpServ * srv=new FtpServ(...) is de fact in setup, but on stand alone function.

wchpikus, 2018-11-22:
but it must be below spiffs.begin() so this is not good to move there?

Monarch73, 2018-11-22:
follow the way I did it....

FtpServ *ftpServer;  <--- this goes below the include lines

ftpServer=new FtpServ(....
ftpServer->begin()           <---- these two lines in the setup

Monarch73, 2018-11-22:
I am suspecting that the ftpServer instance gets picked up by the garbage collector....

wchpikus, 2018-11-22:
i test it now..sorry, but after upgrade sdk compilation was terrible slow:(

Monarch73, 2018-11-22:
I could guide you through the installation of visual studio community 2017 and Visual Micro Plugin. Its a really comfortable development environment and its free

wchpikus, 2018-11-22:
yes...i hate arduino..
Eclipse of visual studio..or some normal editor:)

Monarch73, 2018-11-22:
https://visualstudio.microsoft.com/de/thank-you-downloading-visual-studio/?sku=Community&rel=15
 and
https://www.python.org/downloads/release/python-2715/

these are the two things, that you need to install....

Monarch73, 2018-11-22:
make sure, you tell visual studio setup to install the c++-development component

wchpikus, 2018-11-22:
VC++ 2017 version 15.7 v14.14 toolset?

Monarch73, 2018-11-22:
in the workload menu of the setup, just select  
[image]

wchpikus, 2018-11-22:
https://www.visualmicro.com/page/esp8266.aspx
ok tnx, and this for esp

wchpikus, 2018-11-22:
Could i import project from arduino to visual?
Offcource ino to c rename..

Monarch73, 2018-11-22:
to install the arduino sdk for ESP later via git:

- create some empty directory
- create a directory called "hardware"
- inside hardware, you create another directory "esp8266"
- inside "esp8266" directory you execute git-command (git bash) : git clone --recursive https://github.com/esp8266/Arduino.git esp8266 
- switch to release code by cd-ing into esp8266 and run: git checkout tags/2.4.2
- In it, you'll find a directory called tools...
- execute "python get.py" in command line

Monarch73, 2018-11-22:
unfortunatly, there is no easy way to import arduino stuido project into visual studio....you will have to create a new visual-micro project in Visual studio and copy and paste the files manually

wchpikus, 2018-11-22:
Sure, it is no problem to copy;)
I sow that is possibility to debug code,is it true?
Could we talk to priv, my @ wchpikus@gmail.com.
Big tnx man for help, is it hard to find someone help with esp;)

Monarch73, 2018-11-22:
I just sent you a mail...

wchpikus, 2018-12-03:
to gmail?

wchpikus, 2018-12-03:
i tried on ccp do this:
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI *u8g2;

void init_screen(){
  u8g2 = new U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 15, /* dc=*/ 2, /* reset=*/ 5);
  u8g2->begin(); // use -> instead of . to access methods now
}
and the h file:

extern U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI *u8g2;
void init_screen();

and:
init_screen();  in main file..
and first function u8g2.setContrast(backligth); compile error..

request for member 'setContrast' in 'u8g2', which is of pointer type 'U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI*' (maybe you meant to use '->' ?) 

Could you tell me hot to declare this ssd on other files in project?

Monarch73, 2018-12-03:
`u8g2 = new U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=/ 15, / dc=/ 2, / reset=*/ 5);`
I assume you meant:
 
`u8g2 = new U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI(U8G2_R0, /* cs=/ 15, / dc=/ 2, / reset=*/ 5);`

wchpikus, 2018-12-03:
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 15, /* dc=*/ 2, /* reset=*/ 5); => ok

u8g2 = new U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI(U8G2_R0, /* cs=/ 15, / dc=/ 2, / reset=*/ 5); =>'u8g2' was not declared in this scope after compilation