nailbuster/esp8266FTPServer #26: Able to download fine but not upload some files

bill-orange, 2019-03-29:
I am running esp8266FTPServer on an ESP8266.  I have used this library for several years without any issues.  I recently had to recompile some old code using the Arduino IDE 1.8.8 plus the newer 2.5 board manager.  I am not sure if my problems relate to the new IDE and board manager, insufficient memory or some other unrelated issue.  Filezilla has gone through several revisions since I used this code last too.

Here's the problem:

Using Filezilla I can download fine.  When I try to upload with Filezillza, many files show a size of zero.  If I try to delete them and go again, I get the overwrite message even thought they should be deleted and I still get the zero size file.  My sketch seems able to do these operations fine, no problems upload, downloading or deleting if done within the sketch.  

Not only can I not isolate the source of the problem, I am not sure how to go about troubleshooting it,  

Thoughts anyone?

stephen-coates, 2019-04-05:
Just loaded this today and had the same issue, any attempt to upload results in a zero byte file as does renaming  an existing file. After a bit of experimentation I point the finger at Filezilla,  WinScp and Windows 10 explorer both upload and rename files without issue.  Filezilla appears to upload the file as a hidden file with the filename prefixed by a comma e.g.  ,<filename> and also creates a file named +r,<length of uploaded file>. Possibly Filezilla is expecting a response from the FTP server before converting these files to the uploaded name but if so it's a piece of FTP arcana that Microsoft and WinScp cheerfully ignore.

tbienvenu, 2020-02-29:
I've the same problem as bill-orange, files are 0 length.
I found that filezilla closed too quickly connection after send data so the ESP see a available connection but data connection status are closed() before the esp test it. 
However data are received by the ESP and data.available() return true on the connection.

bill-orange, 2020-02-29:
Interesting. I finally gave up and switched to CoreFTP which seems to work fine.

Jahencov, 2020-10-29:
Now I see it is similar issue that I had. 
Some solution applied by me described here:
https://github.com/nailbuster/esp8266FTPServer/issues/33