nailbuster/esp8266FTPServer #32: Migration to LittleFS File System

BOBILLIER, 2020-06-05:
Hi, thanks for library. It's works good for me, but since Esp8266 community Library 2.7.1, SPPIFS seem to depreciate and we are encouraged to migrate to a LittleFS file system. I have tried to convert all SPPIFS references to LittleFS reference in your library. The compilation runs Ok, and I can connect to my FTP server without problem. But when I try to Upload or download one file, I obtain an error and can't do it. For example, when I want to show root directory, I get "425 No data connection MLSD". Is-there any chance than you upgrade your library to be compatible with LittleFS file system ?
Best Regards
Eric

BOBILLIER, 2020-06-09:
I Have try again and now i can place File on FTP server, but strangely the first caracter of file name are removed. (L123.txt became 123.txt) and after transfert, i can't do nothing with file (like Show or download it).

DmytroKorniienko, 2020-06-15:
> I Have try again and now i can place File on FTP server, but strangely the first caracter of file name are removed. (L123.txt became 123.txt) and after transfert, i can't do nothing with file (like Show or download it).

Comment out all "fn.remove(0, 1);" lines in the ESP8266FtpServer.cpp

BOBILLIER, 2020-06-16:
You are right .. I have comment those lines  and now all seem work fine. I send you one copy of my modified library. I don't have test with esp32 but work fine for dowload and upload on esp8266.
I have add one method to know if ftp client is connected.
[esp8266FTPServer_LittleFS.zip](https://github.com/nailbuster/esp8266FTPServer/files/4785503/esp8266FTPServer_LittleFS.zip)

closed 2020-06-16 by BOBILLIER